makerbase-mks/MKS-SERVO42C #2: What driver does the 42C use?

vader333, 2021-11-15:
Hi, I'd like to know what 4 half bridge with 8 mosfet means. Does it have a datasheet?

KaruroChori, 2021-11-16:
I guess it means that the microcontroller in the board is directly responsible of the generation of the driving signals for the motor (as well as the integration of the signals coming from the feedback loop). Usually if you had something like a TMC2209, a similar driving circuit would be integrated internally and only high level features like performing a step with a certain direction are exposed.

VitRuzicka, 2021-11-19:
> Hi, I'd like to know what 4 half bridge with 8 mosfet means. Does it have a datasheet?

I would also love to have the option of datasheet or PCB files, so I could make my own light version with CAN bus.
Vit

dzid26, 2022-02-23:
> > Hi, I'd like to know what 4 half bridge with 8 mosfet means. Does it have a datasheet?
> 
> I would also love to have the option of datasheet or PCB files, so I could make my own light version with CAN bus. Vit

Look at S42Bv2 if you would like something simple with CANbus