makerbase-mks/MKS-SERVO42C #4: Klipper support

viappidu, 2021-11-30:
Wouldn't be nice to see direct Klipper support a-la https://github.com/Klipper3d/klipper/blob/master/klippy/extras/servo.py?

KaruroChori, 2021-11-30:
What do you mean with "klipper support"? I am using it in klipper, and it does work just fine.
If there is something missing it is probably klipper itself which should be integrating it. Unless you mean you would like to be able to run klipper inside this device.

viappidu, 2021-11-30:
>  Unless you mean you would like to be able to run klipper inside this device.

Exactly!

KaruroChori, 2021-11-30:
I would like this to be possible as well, however this board is responsible for the low level operation needed to control the motor as well as the feedback loop. It has no external driver unlike previous models.
This would require some serious dedication from klipper's developers as such features are not internally supported and I do not think they will ever be if we consider their limited applicability and the fact servos are not very popular in that community.

The best makerbase could do to support this and other projects, is to publish the datasheet with the pinout and possibly the source code of the firmware running in the servo controller. I really hope they will. Any further action (and effort) would still be on the klipper side.

AndreyShpilevoy, 2021-12-02:
> I would like it to be possible as well, however this board is responsible for the low level operation needed to control the motor as well as the feedback loop. It has no external driver unlike previous models. This would require some serious dedication from klipper's developers as such features are not internally supported and I do not think they will ever be if we consider their limited applicability and the fact servos are not very popular in that community.
> 
> The best makerbase could do to support this and other projects, is to publish the datasheet with the pinout and possibly the source code of the firmware running in the servo controller to be used as reference. I really hope they will. Any further action (and effort) would still be on the klipper side.

Hi all,
Sorry for some offtopic,
@KaruroChori KaruroChoribut can you share the specific settings that are you using in Klipper for this board?
Or it's just plug-and-play?

KaruroChori, 2021-12-02:
Plug and play after the initial configuration directly on the servo. You may have to change some parameters on the configuration file due to the microstepping and the direction in case, but aside from that it has the same signal interface of any other driver.

If you want to connect the serial and control it that way I have no idea on how to do that, they only recently released a client for windows.

viappidu, 2021-12-02:
@KaruroChori, I don't agree with you.
It's my belief Makerbase will benefit greatly reaching a huge user base if they decide to add a firmware running klipper natively.

Personally I am going to install Klipper for the first time and I would have no idea where to look for code implementation but I don't think the guys at Makerbase are useless as I am...

KaruroChori, 2021-12-02:
@viappidu klipper is being used by a minuscule proportion of printers world-wide. While it got a lot of traction over the past few months, the vast majority of printers are still based on marlin, reprap or some proprietary solution. Just because there is plenty of videos online of people appreciating this platform it does not mean that most people are willing to go down this rabbit-hole, and as far as I know there is virtually no printer directly sold with klipper inside. They are all portings and customized solutions handled by communities. Why should it be any different this time?

Even more so considering the small number of people buying this product. It is already a niche application, how could you expect a company to devote money and time to address the desires of a niche in a niche?

Furthermore, I should be re-iterating the bigger elephant in the room which I suggested before: Klipper is not ready to support the functionality provided by this board. It would require some major additions on the "raspberry-side" so-to-say before this servo could even be functional in the way you envision. And makerbase has no say in how klipper is being developed, it is well beyond their control. Even assuming they would write all this code on their own, there is no guarantee it would be merged in the main repository.

viappidu, 2021-12-02:
https://github.com/makerbase-mks/MKS-Monster8/tree/main/klipper%20firmware
How does this apply to your idea of a niche?

Re the ability of klipper to support the functionality of the servo... I see no reasons why @KevinOConnor would not implement it if Makerbase write those functions. Why would he not??

KaruroChori, 2021-12-02:
Can we appreciate that:
- every printer has a main board inside
- providing a precompiled image of the firmare from an officially supported architecture and a configuration file is not much effort at all since you can do it yourself in few minutes at most?
On the other hand, virtually nobody uses servos in this context. How would you call it?

For many reasons:
- the quality of the code is not up to the standard
- they may feel this feature is not very important and not worth the effort to test it properly
- and much more.

viappidu, 2021-12-02:
Oh boy!
I'm just glad you don't take decisions for Makerbase!

KaruroChori, 2021-12-02:
Clearly I do not, and this was not my intent. I was just commenting on what is likely going to be the outcome of your request considering the context in which it was made, so that you can calibrate your expectations in case you planned on doing something specifically with this product.

I am the first wishing for these features to be there or implemented in the near future, but I am also quite realistic. I am just reporting these "issues" because I have seen similar cases before, and your enthusiasm for klipper might be biasing your point of view a bit.

Cheers
These are just my 2 cents. Make what you want out of them.

sarumpaet, 2022-04-11:
Running the microcontroller part of Klipper on the 42C would be cool but users would still have to get some hardware to connect the RPi/PC with the 42C so not sure what the immediate win there is apart from some theoretical performance gains.

It'd be nice to have a mode that allows bidirectional communication via the EN/STP/DIR pins though so that a) no extra connection to the main control board is needed and b) Klipper running on the main control board can access (control+read) 42C without the "four extra lines for UART" kludge. This would bring 42C up to par with TMC22xx UART/SPI modes.

Long term it'd be super cool to have a 42D that has Klipper compatibility directly, offering a USB connection, and getting powered via USB Power Delivery. That'd mean you could plug the 42D into some PD enabled USB hub that's connected to the Klipper host and you're done. No further specialized hardware needed. Now only those USB enabled hotbeds and hotends are missing, and you have a 3D printer without any dedicated control board.

xadonxander, 2024-01-27:
Ya'll are way over thinking it. The servo42d firmware needs to be released and the can bus interpreter needs to be reconfigured to follow the klipper can bus protocol. There are a couple IO on the board that could be utilized int he same way. It would be easy with the firmware but MKS refuses to release. I would love to have can bus connectivity within klipper to these motors.

gretel, 2024-01-30:
> Ya'll are way over thinking it. The servo42d firmware source needs to be released and the can bus interpreter needs to be reconfigured to follow the klipper can bus protocol. There are a couple IO on the board that could be utilized int he same way. It would be easy with the firmware but MKS refuses to release. I would love to have can bus connectivity within klipper to these motors.

think you are overthinking things cause the firmware is not released and eventually will not.

xadonxander, 2024-01-30:
> think you are overthinking things cause the firmware is not released and eventually will not.

Yeah, that's why I just returned all 4 of the servo42d's I ordered. Completely useless junk as they are.