makerbase-mks/MKS-SERVO42C #5: Smooth acceleration

DenisD3D, 2021-12-03:
Hi,
We found in the documentation that the MKS-SERVO42C can control the accelaration of the motor : https://github.com/makerbase-mks/MKS-SERVO42C/wiki/Menu-instruction#parameter-configuration-suggestions
But setting the Mplyer to disable and the acc to 286 or 1042 don't change anything to how the motor accelerate
How should we setup our motors to have a progressive acceleration/deceleration ?
Thanks
Denis

gjyskyblue, 2021-12-04:
We achieve progressive acceleration/deceleration by changing the signal frequency of the STP pin. 
Increasing the value of ACC can make the motor respond to a higher signal frequency.

DenisD3D, 2021-12-04:
I forgot to mention we are using the UART protocol to control our motors and so can't control the step pin

gjyskyblue, 2021-12-05:
I'm sorry, the UART can't setup the acceleration/deceleration now.
We will provide PID and ACC parameter settings for UART in the next version.

DenisD3D, 2021-12-18:
Hi, is there any ETA for the new version of the firmware ? Could we except it to be available in january ?
Thanks
Denis

cricket2009, 2021-12-21:
If you need to set the parameter of PID  from UART , you can contact the AliExpress Customer Service，and you will be upgraded to the new  firmware.