makerbase-mks/MKS-SERVO42C #7: Enable / Disable Pin affects not only the coils but also the encoder!

bodyhey, 2021-12-30:
If you use the serial command 8: "e0 f3 00" to disable the coils (to move the shaft freely), also the output of the encoder doesn't change any more. That is quite unfortunate, if you want to record a passive movement. Is there a possibility to just deactivate the coils and still get position values?

nyinyinyanlin, 2021-12-31:
I find this EN status affecting encoder really counter-intuitive as well. There should be a separate feature to enable/disable coil holding. I believe releasing the firmware public will help the development and implementation of useful features faster and comprehensive.

nyinyinyanlin, 2021-12-31:
@bodyhey I have observed a rather strange behavior. If you send a "ex fd xx xx xx" command and send a EN disable command immediately after, the motor shaft status became blocked and the coil is disabled. BUT........ you can still send the motor shaft angle read command and get the updated angle when the shaft is moved by hand. (Encoder value and error angle aren't updated though.) May be a bug. Once you set EN to enabled again, current angle is read as 0 degree instead.

@Huang-jh can you please implement a command to enable/disable coil without affecting the encoder and a command to return to 0 degree position set through menu button?

Thank you very much!

cricket2009, 2022-01-06:
@bodyhey@nyinyinyanlin Thanks for your proposal,we will implement a command to enable/disable coil without affecting the encoder.

cricket2009, 2022-01-06:
@nyinyinyanlin "command to return to 0 degree position" , Is the "0 degree position" preview set by menu "Set 0"?

nyinyinyanlin, 2022-01-08:
@cricket2009 My bad, I made it sound confusing. What I meant was a serial UART command to return the shaft to 0 degree position. The motor shaft automatically return to 0 position on power up, if a 0 position is set by menu. The same functionality through UART command implemented will be great! I can implement that in my own program code by keeping track of current shaft angle and rotate to 0 angle position. But it would be better if the MKS-SERVO42C has a built-in UART command to rotate back to 0 position.

nyinyinyanlin, 2022-01-08:
@cricket2009 And additionally it would be great if RESPONSE packets have a packet start identifier byte and a checksum byte.

cricket2009, 2022-01-09:
@nyinyinyanlin  I understand now, thank you very much.  we will implement a command to  return the shaft to 0 degree position by serial UART command. 
It is also a good idea to add a header and checksum to the response frame.

bodyhey, 2022-01-09:
For the next firmware version: I found a strange behaviour when moving a motor at a certain speed via UART: The effective speed of the shaft drops to zero when using HEX 60 at a stepper with the ID E0. The same happens with the setting "61" to a stepper E1, "62" to a stepper with the ID E2 etc.

cricket2009, 2022-01-10:
@bodyhey  do you mean:
if the motor ID is "E0"  , send command "E0 60" by Uart,the Motor speed will be set to 0?
if the motor ID is "E1"  , send command "E1 61" by Uart,the Motor speed will be set to 0?
...
Maybe is a bug, we will test it, thank you!

bodyhey, 2022-01-11:
@cricket2009 : Yes, exactly.