makerbase-mks/MKS-SERVO42C #8: Enable slower speeds

bodyhey, 2022-01-21:
Even in "lowest gear" - with the speed setting "0x00" the SERVO42C turns relatively fast (if you use it for animatronic): Could you implement some kind of "slow mode" command into the next firmware, scaling down the speed by a factor of 16-128?

bodyhey, 2022-01-22:
@cricket2009 Could you give any information about a possible date for a new firmware? And will it be possible to update already existing SERVO42Cs or will the new version only be shipped with new ones? 
The background of my question: If there is no timeline for a relase, I will have to aim for a unelegant solution: timing belts with a reduction of 1:3 - 1:6. This will only add unnecessary complexity and costs to my projects, a simple software solution (e.g. new firmware) would be much more elegant. 
I like the SERVO42C very much and already  bought 13 of them. The fields, where they could be used, could be much, much broader with a little more flexibility in respect to speed. Absolutely perfect woould be a range from 0x0000 - 0xFFFF corresponding to 0 RPM up to the maximum possible 1.000 RPM.
If there are no resources to programm all our wishes, maybe Makerbase could again discuss, to make the firmware open source and let the community do the work?

cricket2009, 2022-01-24:
@bodyhey  Thanks for your suggestion！
we will implement "slow mode" command by uart, the maximum speed will be set to 1.000RPM. 
The new firmware will be released in February or March 2022.

bodyhey, 2022-01-25:
@cricket2009: Thank you very much!!! These are phantastic news!