makerbase-mks/MKS-SERVO42C #9: Reverse Lookup Error

nyinyinyanlin, 2022-01-27:
I am getting reverse lookup error when calibrating. FYI, I am driving a geared yaw mechanism with the stepper and the MKS-SERVO42C magnetic sensor is placed just above the yaw axis, not on stepper.

Why am I getting this? What should I do? Thank you very much!

cricket2009, 2022-01-28:
1. The distance between the magnet and the encoder should be 0.5~3.0mm.
2. The motor rotates 200 steps (1.8 degrees) or 400 steps (0.9 degrees) for one revolution.

nyinyinyanlin, 2022-01-28:
@cricket2009 Thank you very much for your quick response. The distance is in correct range. 

I'm using a belt pulley system to drive the yaw mechanism. The servo42c module is not attached to stepper, but to the yaw axis of the pulley/gear of the main body. The gear ratio is 3:1. So I guess the stepper will have to rotate 3 revolutions to rotate the yaw axis one complete revolution. 

Anyway to bypass or modify this calibration to suit my need? Because I would to detect the rotation angle and error on the main yaw axis, not on the stepper. Even if the stepper did not slip, there's a chance of the gear/belt slipping. I'm using this module to write a paper on development of a measurement system with yaw/pitch pan/tilt mechanism. And as the deadline is really close, I would appreciate much if you can give me a quick fix/tweak to this. I'm driving the module from UART. Major functionality required is to read the encoder value to detect the angle of the yaw axis and stepper driver to drive the stepper. This is the bare minimum need in my case. I can implement error angle finding and position keeping in my code.

sarumpaet, 2022-03-11:
Just for reference in case anybody web searches this in the future:
I got a **Reverse Lookup Error** when trying calibration (and the stepper would only jiggle a bit back and forth instead of slowly rotating) for a 42SHD0217 NEMA17 stepper which uses a **pinout that is not what SERVO42 expects**.

https://github.com/makerbase-mks/MKS-SERVO42C/wiki/Makerbase-SERVO42C-Closed-loop-Stepper-Motor--Manual%EF%BC%881%EF%BC%89-Install-the-motor has a section concerning pinouts but most other installation descriptions don't.

For me, just switching the central two wires fixed the issue.

Note that I got a "Reverse Lookup Error" and _not_ the "Phase Line Error" mentioned in the manual.

westonforbes, 2025-10-30:
> Just for reference in case anybody web searches this in the future: I got a **Reverse Lookup Error** when trying calibration (and the stepper would only jiggle a bit back and forth instead of slowly [...]

@sarumpaet, you're my hero for the day. Cross wires here too. Thank you so much!