makerbase-mks/MKS-SERVO42C #13: Motor slammed into the side of the printer - servo board does not work anymore?

Jim-bee, 2022-02-11:
Hello,

The motor with servo board installed was working fine.  The motor on its own changed direction and the carriage slammed into the side of the printer. It stopped there with the motor vibrating and before I could kill the power it just stopped and then nothing.  The the motor is fine but servo board does not work anymore.  There is no OLED display and it now shorts out the 24 V and 3.3V power supplies.  Is there any protection built into the servo board so it cannot damage itself or anything else in circuit?  Any thoughts on what may have been damaged on the servo board ie: mosfets, etc?

Thanks,
Jim-bee

cricket2009, 2022-02-14:
I am sorry that this happens, It seems that the mosfets are damaged.
It can be set the "Protect" to "Enable" by the oled screen menu.
if the motor is blocked, the driver board will disable the motor and the mosfet will not be damaged.

Jim-bee, 2022-02-15:
Hello,

Thank you for your reply and information, good to know you have the protect function.  I wish the protect default was enabled.  I would like to try to fix this board.  Can you please let me know the manufacture and partnumber (specifications) for the mosfets and the H-bridge IC’s.

I have another issue / question.  In my testing (using 64 steps) doing circles the circles are not round when using your servo board.  If I do the same test without your servo board (just the motors) the circles are perfectly round.  I created a jig to exactly center the magnet on the motor shaft (oled displays 0.0 error).  Is there an adjustment to fix this issue.

Best Regards,
Jim

Jim-bee, 2022-02-26:
HI cricket2009,

I tried enabling the "Protect" and tested the motor does not stop it still sits at the end stop vibrating.  This does not seem to work as you mentioned.

cricket2009, 2022-03-02:
Hi Jim-bee  
1. the mosfet is CJ3400
2. After setting the "Protect" by the menu, enable the driver board, you can force the motor to rotate by hand, the driver board will fail when the torque increases.
 Best Regards!