makerbase-mks/MKS-SERVO42C #15: Circles are not round

Jim-bee, 2022-02-16:
Hi, 

I have another issue / question.  In my testing (using 64 steps) doing circles the circles are not round when using your servo board.  If I do the same test without your servo board (just the motors) the circles are perfectly round.  I created a jig to exactly center the magnet on the motor shafts (oled displays 0.0 error).  Is there an adjustment to fix this issue.

Thanks,
Jim-bee

cricket2009, 2022-02-21:
Sorry for this issue,
You could try the following configuration :
MPlyer : Disable
ACC : 412

Pastitas, 2022-02-21:
could you explain why that's the cause or how does it generate said issue? Just for future reference and so people can diagnose and fix these kinds of issues themselves i think it might be valuable.

Jim-bee, 2022-02-26:
Hi cricket2009,

I tried your settings and they do not seem to help with the circles.  My test is using a pen and I draw 3 different diameter circles in the CW direction then in the CCW direction and the circles do not line up throughout the circumference of the arc.  If I run this same test with just the stepper motors (no servo board) the circle overlay each other perfectly.  Please see the attache photo.
[image]

Pastitas, 2022-03-01:
@Jim-bee By any chance do you have the jig available for download? I'd like to try and replicate your results and I don't think my magnets are spot-on

GeorgScha, 2023-03-02:
> Hi,
>
> I have another issue / question. In my testing (using 64 steps) doing circles the circles are not round when using your servo board. If I do the same test without your servo board (just the [...]

I had the same issue with Servo42C ( Board lettering says Servo42C V1.0, I don't know the Firmware version)
I tested Servo42A and Servo 42B Boards, they work fine. ATM Servo42B is installed and works. Only thing is, Servo42C board was very quiet, A and B make some small Motor noise. I'd like to go back to Servo42c for this reason.
For your Information: I just switched the PCB Boards, Motor and installed magnet did stay the same! So I can say for sure that it is only the C board causing this problem. Tested many different Settings, did not help. Did anyone solve this issue?
Be careful, if you switch to Servo42B, the pins of the Driver Adapter PCB ( 6 Pin Cable conncetion) are different. So go sure to switch both components.

Georg

Ingrow-Cloud, 2023-03-04:
Hey! I do have the same problem. Is there any fix now? 
Have a great Day!

Jim-bee, 2023-03-29:
Hi GeorgScha & Ingrow-Cloud,

Good the see I’m not the only one seeing this issue with circles.  I also tried using the 42C board and disabled the servo mode - so just using the board as a standard stepper driver and the circles overlayed perfectly.  
I have not been able to solve it.  I believe Makerbase has a problem with their servo control / math in their firmware because, as shown in my photo, parts of the circle overlay perfect and other parts of the circle do not.  I wish they
would either fix the issue or release the source code.

Jim

GeorgScha, 2023-08-16:
Hi Guys,Today I installed the S42C from Bigtreetech (BTT) . This one does not have the oval issue and runs very quiet. Until now, looks great. Will run it for many hours the next time, as soon as I have print-Jobs for it :-) 
The only disadvantage I still see in all closed loop boards ( tested 4 different ones now) is the low max.speed. With a 11mm Diameter pulley on the motor the max. Speed is 250mm/s . Higher speeds lead to ugly, jerky movement with all boards.
Georg

GeorgScha, 2023-08-16:
Small summary ( my experiences!) :
MKS S42A: runs fine, but noisy
MKS S42B: runs fine, but noisy
MKS S42C: oval circles issue, could not solve it. Runs smooth
Bigtreetech BTTS42C:  runs fine and smooth until now.