makerbase-mks/MKS-SERVO42C #17: Sensorless homing feature

CapnBio, 2022-02-25:
Hey all!

I was wondering if it is at all possible to do sensorless homing. For example on the x and y axis. I was wondering if it is at all possible to do that with UART and what not. I already have the kits, just want to save the hassle of even buying end stops.

mozg4D, 2022-10-24:
Only 6 wires from driver to board - spep, dir, ebnable, ground, vcc, vin. So it seems sensorless homing not possible. This is a big problem for me as my printer has no endstops and has no mounting holes for endstops, it was designed for sensorless homing only. aww(
You may not use homing at all - simply hand move axes to zeros and execute G92 X0 Y0 command, but in that case there's a bug in marlin with "auto bed leveling". It does not work without homing X Y. In my case "auto bed leveling" must be performed

DanielMarcato, 2024-01-05:
can't the locked-rotor protection function be used for sensorless homing? sounds like this is what it is made for - but I have not tried it yet

Lucipuki2, 2025-12-02:
@DanielMarcato Have you had the chance to try it yet? Im building a printer where senserless homing is required.