makerbase-mks/MKS-SERVO42C #18: "Wishlist" for next firmware

bodyhey, 2022-02-28:
Dear team of makerbase-mks,

[cricket2009](https://github.com/cricket2009) suggested, a new firmware for the SERVO42C will be released
in february or march 2022.

Some new features had been discussed to make the SERVO42C an interesting solution for even more applications:

- A Slow-Speed-Mode, where via UART the values 0x0000 - 0xFFFF correspond to 0 RPM up 100 RPM.

- An UART command to enable/disable coil holding with the encoder values still to be updated (use case: if you want to record a passive movement). The problem right now: If you use the serial command 8: "e0 f3 00" to disable the coils (to move the shaft freely), also the output of the encoder doesn't change any more. 

- An UART command to return to 0 degree position,  set by menu "Set 0"

- An UART command to execute the "Set 0"

- A little bug fix: I found a strange behaviour when moving a motor at a certain speed via UART: The effective speed of the shaft drops to zero when using HEX 60 at a stepper with the ID E0. The same happens with the setting "61" to a stepper E1, "62" to a stepper with the ID E2 etc.

- An UART command to drive the stepper to an absolute angle position at a certain speed (see also "Slow Speed Mode"). As position value, the value from the existing command 3 : e0 36 (read the angle of the motor shaft) would be perfect. e.g. something like "e0 f9 YY YY XX XX XX XX with direction / speed YY YY and angle (int32_t XX XX XX XX)

Do you think, these wishes are realizable? The Slow-Speed-Mode would be the most urgent, as for every other problem there are more or less elegant workarounds.

All the best & let me know if I could support you ...

Martin

P.S.: A very cool feature would be a "force feedback control"-mode where you just connect two SERVO42c via RX/TX TX/RX without a microcontroller between. Something like that: https://www.youtube.com/watch?v=xTlv1rPEqv4

cricket2009, 2022-03-01:
@[bodyhey] Some of the above functions have been implemented, and before March 15th, I will list the new functions implemented.

cricket2009, 2022-03-01:
[image]
the new firmware RPM calculation :
for 1.8 degree motor:
RPM = speed *500*60/(Mstep*200)
for 0.9 degree motor:
RPM = speed *500*60/(Mstep*400)
for example:
speed=1, Mstep=150, 1.8degree 
RPM = 1*500*60/(150*200) = 1

cricket2009, 2022-03-01:
the new firmware RPM calculation :
for 1.8 degree motor:
RPM = (speed x 500 x 60)/(Mstep x 200)
for 0.9 degree motor:
RPM = (speed x 500 x 60)/(Mstep x 400)
for example:
speed=1, Mstep=150,1.8degree 
RPM = 1x500x60/(150x200) = 1

cricket2009, 2022-03-01:
[image]
the new firmware RPM calculation :
for 1.8 degree motor:
RPM = (speed x 500 x 60)/(Mstep x 200)
for 0.9 degree motor:
RPM = (speed x 500 x 60)/(Mstep x 400)
for example:
speed=1, Mstep=150,1.8degree 
RPM = 1x500x60/(150x200) = 1

uboatsalamander, 2022-03-03:
Nice news, what is the method of uploading firmware, UART?