makerbase-mks/MKS-SERVO42C #20: About the new firmware V1.1

cricket2009, 2022-03-09:
The V1.1 firmware has made the following improvements：
Screen menu :
1. Added automatic zero return option "Goto 0"
2. Added the option "Restore" to restore the default parameters
Serial command:
1. All screen menu functions are realized;
2. The check value is added to the sent and received data;
3. PID, acceleration (ACC), maximum torque (MaxT) parameters can be set;
4. The speed of the serial control motor can be set arbitrarily (0.6RPM ~ 1200RPM)
Remaining problem:
1. An UART command to enable/disable coil holding with the encoder values still to be updated.

cricket2009, 2022-03-09:
about the remaining problem:
1.  after enable the driver by command "e0 f3 01" ,it can be read 
		 a. the encoder value("e0 30")
		 b. the number of pulses received("e0 33")
		 c. the angle of the motor shaft("e0 36")
2. If the motor stops at position A, and release the motor by command "e0 f3 00" ,
	 the encoder value still updating, 
	 but it does not receive the pulses,  
	 and can't calculate the angle of the motor shaft.
3. turn the motor to position B by hand, and enable the driver again. 
	If the pulse and angle parameters are not set correctly or cleared to zero, 
	it will automatically return to position A.
	So, I had to clear the parameter to 0 when re-enabling the driver.

MakeTeachRepeat, 2022-03-15:
> The V1.1 firmware has made the following improvements： Screen menu :
> 
> 1. Added automatic zero return option "Goto 0"
> 2. Added the option "Restore" to restore the default parameters
>    Serial command:
> 3. All screen menu functions are realized;
> 4. The check value is added to the sent and received data;
> 5. PID, acceleration (ACC), maximum torque (MaxT) parameters can be set;
> 6. The speed of the serial control motor can be set arbitrarily (0.6RPM ~ 1200RPM)
>    Remaining problem:
> 7. An UART command to enable/disable coil holding with the encoder values still to be updated.

Is the firmware possible to flash onto board? What is needed?

viappidu, 2022-03-15:
I might be missing something but...
Where is the firmware??

bodyhey, 2022-03-24:
Dear cricket2009,

The features of the new firmware are very nice, thank you!!! I wish you all the best, to find a solution for the "enable/disable coil holding" problem.
I have the same question as viappidu: Where can I download the new version and how can I flash the SERVO42C?

uboatsalamander, 2022-04-03:
Hello there was an update on the page https://github.com/makerbase-mks/MKS-SERVO42C/releases/tag/MKS-SERVO42C-V1.1, however the attached zip archive contains no firmware. When is it planned to release firmware for the existing owners of the Servo42C? Thank you

dbarvik, 2022-04-04:
Hello I am also asking for the firmware, I have read capabilities of the motor here on github and when I purchase this Servo42C, it doesn't support all functionality. I cannot control steps of rotation by uart, which was the reason to buy this device.

I would really appreciate if you provide the firmware for the Servo42C. Thank you

bodyhey, 2022-04-15:
Here are the serial commands for v1.1: 
https://chowdera.com/2022/04/202204051042164971.html

fca1, 2022-04-21:
Hi, We are waiting to upgrade our MKS42C already bought. Please, could you give us a date of availability of firmware V1.1 ?

kutukvpavel, 2022-07-04:
Please upload at least v1.1 binaries (and flashing instructions, I assume the HC32L130 can be put into DFU mode or something?). I have some 42C boards with v1.0 firmware I'd like to upgrade.

PeterMobo, 2022-07-23:
https://programmersought.com/article/91177446965/
https://www.programmersought.com/article/84674105265/
These two links show to program the controller used on the MKS42 module. You only need a cheap USB-UART converter working with 3.3 Volt.
The needed connection is shown at the attached picture. 
[image]
Additionally you need to pull the BOOT0 pin high (3,3V). 
Disclaimer: All information were not found out by myself and are just borrowed from user 'bodyhey' from the MKS discord server. You will find more information there on the discord server of MKS products.

Now the big 'BUT': I looks like the firmware or µC are protected. It seems not to be possible to read out firmware 1.1 from one board and to burn it into another board. If MKS will not provide any update procedure and refuse any communication in this regard, i must and will advise others against this product in the future. The MKS42c It is improving through the feedback of us (the customers/users) but then we are not able to participate and have to buy new hardware instead. Sorry, this is year 2000 style and not acceptable. Todays world is different and this shall not be supported anymore.
I can understand MKS that they want to protect their software in some way, but you (MKS) will have to find a solution here!! Simply playing dead on this issue is not a solution!

gaby64, 2022-10-14:
is mks listening?

crstrand, 2022-11-25:
Please please please please release the 1.1 binary

crstrand, 2022-12-03:
STM32L010K8 is almost pin compatible...