makerbase-mks/MKS-SERVO42C #23: v1.1 - when moving slowly, load on the shaft leads to fast spinning

bodyhey, 2022-06-04:
When you are in UART mode and set the motor in lowest gear (command "e0 f6 01 d7") and then give some resistance against the shaft, sooner or later the motor will start turning in high speed.
Interestingly, on the display the position (degrees) still changes slowly.
This is a very dangerous bug, because it has the potential to destroy the machine, the motor is used for.

gjyskyblue, 2022-06-05:
Sorry for the bug, we will test and fix it.

bodyhey, 2022-06-05:
@gjyskyblue Thank you very much! Will you provide the updated firmware here on github?

bodyhey, 2022-09-25:
@gjyskyblue @makerbase-mks Do you have any new information an this problem? Were you able to replicate the error? I can't proceed with my project, if there is no solution - even if it means, I have to buy new boards with firmware 1.2.

mboremski, 2023-02-20:
Will there be a fix for this issue?
I bought one of those 2 weeks ago..
It states Firmare 1.1.2

While using uart nearly every 3rd command leads to fast spinning!