makerbase-mks/MKS-SERVO42C #24: Is there any example code for Arduino?

csvke, 2022-06-22:
Would like to use this driver with Arduino.

Is this driver based on some other driver that is available with example code with Arduino libraries? The resources in the user manual is very limited.

lundsimon, 2023-07-27:
I was sad I had to figure it all out and then discovered that the motor doesn't save it's position with power off apart from the zero command. I would be so much better if it kept it's position without having to home... anyway, hope this helps. Written for a nano, hence the software serial. Keep the serial cable away from the motor or you get interference.

#include <SoftwareSerial.h>

#define rxPin 4

#define txPin 3
SoftwareSerial MotorSerial =  SoftwareSerial(rxPin, txPin);

unsigned long last = 0UL;      //Set up loop timing
const byte motorPos[]= {0xe0,0x36,0x16};

const byte motorMSGcalibrate[] = {0xe0,0x80,0x00,0x60};     //Message to calibrate motor (WRITE)

const byte motorUARTmode[] = {0xe0, 0xf3, 0x01, 0xd4};       //Put motor in UART command mode

const byte MotorMoveposition[] = {0xe0, 0xfd, 0x86, 0x00, 0x00, 0x0c, 0x80, 0xef};  //motor rotates forward 360° (1 time) in speed 1
int MotorMovepositiovalid[] = {0xe0,0x01,0xe1};                                     //Message to confirm successful move
const byte MotorMoveEnable[] = {0xe0,0x01,0xe1};  

void setup() {
  Serial.begin(9600);
     pinMode(rxPin, INPUT);

    pinMode(txPin, OUTPUT);
    
 MotorSerial.begin(38400);   
  delay(100); 
 MotorSerial.write(motorUARTmode, sizeof(motorUARTmode));
   
   

}

void loop() {
  float rps=.025; //2=250 1=125 .5=60 .025=40 sec total.5/60=1s exp
 // Vrpm = (speed × 30000)/(Mstep(256) × 200)(RPM) 1.8°motor
 //(rps*60*(256*200)/3000=speed
 int msteps=128;
byte spd=(rps*60*msteps*200.0)/30000.0; //cant be more that 125 or runts int dir bit
  Serial.print("spd=");
   Serial.println(spd,DEC);
  
   delay(100);
 int angle=listenAndPrintPosition();
if (angle>=90){
   makeMove( 180/(1.8/msteps),spd,  0);
   
    
    delay(int(1/rps*1000.0*1.20));
    listenAndPrint();
     delay(1000);
}
else{
  makeMove( 180/(1.8/msteps),spd,  1);
    delay(int(1/rps*1000.0*1.20));
   listenAndPrint();
  delay(1000);
}
   /*
  Serial.println("\nForward=");
   MotorSerial.write(motorrev360spd1,sizeof(motorrev360spd1));
    delay(5000);
     listenAndPrint();
     Serial.println("\nMotorPos=");
  MotorSerial.write(motorPos, sizeof(motorPos));
  delay(1000);
  listenAndPrintPosition();
        Serial.println("\nRev=");
  MotorSerial.write(motor360spd1, sizeof(motor360spd1));
  delay(8000);
  listenAndPrint();
   Serial.println("\nMotorPos=");
  MotorSerial.write(motorPos, sizeof(motorPos));
  delay(1000);
  listenAndPrintPosition();
  //listenAndPrint();
//MotorSerial.listen();
 
*/
}
void listenAndPrint(){
  
    int numBytes=20; //positin respose
  byte motorData[200];   
    for (int i=0; i<numBytes; i++) {                   //Cycle through each of the bytes in the serial buffer...
        motorData[i] = 0;       
      }
   if (MotorSerial.available() > 0){                    //Check if data is available in the serial buffer
      numBytes = MotorSerial.available(); 
      Serial.println("\nbytes=");//Read the size of the serial buffer
       Serial.println(numBytes,DEC );
      long arraySum = 0;                                 //Reset the arraySum to zero
      long arraytCHK = 0;                                //Reset the array checksum to zero
      for (int i=0; i<numBytes; i++) {                   //Cycle through each of the bytes in the serial buffer...
        motorData[i] =  MotorSerial.read();       
      }
   }
    Serial.println("\nMotorData=");
   for(int x=0;x< numBytes ;x++){
    Serial.print(motorData[x],HEX);
    // Serial.println("\n");
    }
    Serial.println("\nend\n");
  
  }
int listenAndPrintPosition(){
     MotorSerial.write(motorPos, sizeof(motorPos));
  delay(200);
    int numBytes=20; //positin respose
  byte motorData[200];   
    for (int i=0; i<numBytes; i++) {                   //Cycle through each of the bytes in the serial buffer...
        motorData[i] = 0;       
      }
   if (MotorSerial.available() > 0){                    //Check if data is available in the serial buffer
      numBytes = MotorSerial.available(); 
      Serial.println("\nbytes=");//Read the size of the serial buffer
       Serial.println(numBytes,DEC );
      long arraySum = 0;                                 //Reset the arraySum to zero
      long arraytCHK = 0;                                //Reset the array checksum to zero
      for (int i=0; i<numBytes; i++) {                   //Cycle through each of the bytes in the serial buffer...
        motorData[i] =  MotorSerial.read();       
      }
     }
//int motAngle=(motorData[2]*pow(16,7)+motorData[3]*pow(16,6)+motorData[4]*pow(16,5)+motorData[5]*pow(16,4)+motorData[6]*pow(16,3)+motorData[7]*pow(16,2)+motorData[8]*pow(16,1)+motorData[9]);
 uint32_t motAngle = (uint32_t) motorData[4] << 0
             | (uint32_t) motorData[3] << 8
             | (uint32_t) motorData[2] << 16
             | (uint32_t) motorData[1] << 24;
  
  Serial.println("Angle=");
 Serial.print(360.0/65535*motAngle);
 
 /*   Serial.println("PosHex=");
   for(int x=0;x< numBytes ;x++){
    Serial.println(motorData[x],HEX);
    // Serial.println("\n");
    }
    Serial.println("\nend\n");*/
  return 360.0/65535*motAngle;
  }

void  makeMove( uint32_t degree, byte spd, byte dir)
  {
    byte moveDegrees[8];
   moveDegrees[0]=0xe0;//motor addres
    moveDegrees[1]=0xfd; //command id
   moveDegrees[2]=dir<<7|spd;
    moveDegrees[3]=degree>>24;
     moveDegrees[4]=degree>>16;
      moveDegrees[5]=degree>>8;
       moveDegrees[6]=degree;
       //calculate checksum
  moveDegrees[7]=(moveDegrees[0]+moveDegrees[1]+ moveDegrees[2]+ moveDegrees[3]+ moveDegrees[4]+ moveDegrees[5]+moveDegrees[6]) ;
     MotorSerial.write( moveDegrees, sizeof( moveDegrees)); 
    }

jasonmhead, 2023-08-11:
@lundsimon
how was the arduino physically connected to the motor?
did you use a driver board?
what pins where used?

lundsimon, 2023-08-11:
#define rxPin 4
#define txPin 3

 The rx pin 4 on the arduino is hooked up to the tx on the motor, the tx on the arduino to the rx on the motor and the ground on the same header is hooked up btween the two, the 4th pin is not used. On the main motor header you just need to provide power to the motor.
S

lundsimon, 2025-06-21:
> Hi @lundsimon
>
> Thanks for taking the time to paste your code. Its so frustrating that
> they don't even bother to represent Arduino.
> I am battling to read your code as the github format has broken it ..
> any chance you can pastebin it or send it to me ... I am so over this
> mks and the little to know documentation

If you just paste this into the Arduino ide, it should work. I blew up more than one of these boards by crashing the motor and the back emf kills the boards. It seems like if they made it more open, it would be so much more useful:

#include <SoftwareSerial.h>

#define rxPin 4

#define txPin 3
SoftwareSerial MotorSerial = SoftwareSerial(rxPin, txPin);

unsigned long last = 0UL; //Set up loop timing
const byte motorPos[]= {0xe0,0x36,0x16};

const byte motorMSGcalibrate[] = {0xe0,0x80,0x00,0x60}; //Message to calibrate motor (WRITE)

const byte motorUARTmode[] = {0xe0, 0xf3, 0x01, 0xd4}; //Put motor in UART command mode

const byte MotorMoveposition[] = {0xe0, 0xfd, 0x86, 0x00, 0x00, 0x0c, 0x80, 0xef}; //motor rotates forward 360° (1 time) in speed 1
int MotorMovepositiovalid[] = {0xe0,0x01,0xe1}; //Message to confirm successful move
const byte MotorMoveEnable[] = {0xe0,0x01,0xe1};

void setup() {
Serial.begin(9600);
pinMode(rxPin, INPUT);

pinMode(txPin, OUTPUT); 

MotorSerial.begin(38400);
delay(100);
MotorSerial.write(motorUARTmode, sizeof(motorUARTmode));

}

void loop() {
float rps=.025; //2=250 1=125 .5=60 .025=40 sec total.5/60=1s exp
// Vrpm = (speed × 30000)/(Mstep(256) × 200)(RPM) 1.8°motor
//(rps60(256200)/3000=speed
int msteps=128;
byte spd=(rps60msteps200.0)/30000.0; //cant be more that 125 or runts int dir bit
Serial.print("spd=");
Serial.println(spd,DEC);

delay(100);
int angle=listenAndPrintPosition();
if (angle>=90){
makeMove( 180/(1.8/msteps),spd, 0);

delay(int(1/rps*1000.0*1.20)); listenAndPrint(); delay(1000); 

}
else{
makeMove( 180/(1.8/msteps),spd, 1);
delay(int(1/rps1000.01.20));
listenAndPrint();
delay(1000);
}
/*
Serial.println("\nForward=");
MotorSerial.write(motorrev360spd1,sizeof(motorrev360spd1));
delay(5000);
listenAndPrint();
Serial.println("\nMotorPos=");
MotorSerial.write(motorPos, sizeof(motorPos));
delay(1000);
listenAndPrintPosition();
Serial.println("\nRev=");
MotorSerial.write(motor360spd1, sizeof(motor360spd1));
delay(8000);
listenAndPrint();
Serial.println("\nMotorPos=");
MotorSerial.write(motorPos, sizeof(motorPos));
delay(1000);
listenAndPrintPosition();
//listenAndPrint();
//MotorSerial.listen();

*/
}
void listenAndPrint(){

int numBytes=20; //positin respose 

byte motorData[200];
for (int i=0; i<numBytes; i++) { //Cycle through each of the bytes in the serial buffer...
motorData[i] = 0;
}
if (MotorSerial.available() > 0){ //Check if data is available in the serial buffer
numBytes = MotorSerial.available();
Serial.println("\nbytes=");//Read the size of the serial buffer
Serial.println(numBytes,DEC );
long arraySum = 0; //Reset the arraySum to zero
long arraytCHK = 0; //Reset the array checksum to zero
for (int i=0; i<numBytes; i++) { //Cycle through each of the bytes in the serial buffer...
motorData[i] = MotorSerial.read();
}
}
Serial.println("\nMotorData=");
for(int x=0;x< numBytes ;x++){
Serial.print(motorData[x],HEX);
// Serial.println("\n");
}
Serial.println("\nend\n");

}
int listenAndPrintPosition(){
MotorSerial.write(motorPos, sizeof(motorPos));
delay(200);
int numBytes=20; //positin respose
byte motorData[200];
for (int i=0; i<numBytes; i++) { //Cycle through each of the bytes in the serial buffer...
motorData[i] = 0;
}
if (MotorSerial.available() > 0){ //Check if data is available in the serial buffer
numBytes = MotorSerial.available();
Serial.println("\nbytes=");//Read the size of the serial buffer
Serial.println(numBytes,DEC );
long arraySum = 0; //Reset the arraySum to zero
long arraytCHK = 0; //Reset the array checksum to zero
for (int i=0; i<numBytes; i++) { //Cycle through each of the bytes in the serial buffer...
motorData[i] = MotorSerial.read();
}
}
//int motAngle=(motorData[2]*pow(16,7)+motorData[3]*pow(16,6)+motorData[4]*pow(16,5)+motorData[5]*pow(16,4)+motorData[6]*pow(16,3)+motorData[7]*pow(16,2)+motorData[8]*pow(16,1)+motorData[9]);
uint32_t motAngle = (uint32_t) motorData[4] << 0
| (uint32_t) motorData[3] << 8
| (uint32_t) motorData[2] << 16
| (uint32_t) motorData[1] << 24;

Serial.println("Angle=");
Serial.print(360.0/65535*motAngle);

/* Serial.println("PosHex=");
for(int x=0;x< numBytes ;x++){
Serial.println(motorData[x],HEX);
// Serial.println("\n");
}
Serial.println("\nend\n");/
return 360.0/65535motAngle;
}

void makeMove( uint32_t degree, byte spd, byte dir)
{
byte moveDegrees[8];
moveDegrees[0]=0xe0;//motor addres
moveDegrees[1]=0xfd; //command id
moveDegrees[2]=dir<<7|spd;
moveDegrees[3]=degree>>24;
moveDegrees[4]=degree>>16;
moveDegrees[5]=degree>>8;
moveDegrees[6]=degree;
//calculate checksum
moveDegrees[7]=(moveDegrees[0]+moveDegrees[1]+ moveDegrees[2]+ moveDegrees[3]+ moveDegrees[4]+ moveDegrees[5]+moveDegrees[6]) ;
MotorSerial.write( moveDegrees, sizeof( moveDegrees));
}

RobbyGit, 2025-06-22:
Thanks man .. Appreciate it , Again Github has removed or left out parts of the code again. I think you need to put the code in the source code tabs .

byte spd=(rps60msteps200.0)/30000.0; //cant be more that 125 or runts int dir bit

Think the * might be missing from a lot of it . Will work through it.

You Popped mks2 with feedback.. esh!!.. HOW!! thats bad!

 I think i might have picked a bad product for my idea. Ha!

lundsimon, 2025-06-23:
> Hi @lundsimon
>
> Thanks a lot . got it working .. Have you though about doing a library.
> There is one out there but its for Version 1 and not v1.1.2
>
> Again thank you for the time and effort

Great, hope it's useful. I ended up having to sick another encoder on what the motor was moving to know where it was on power up, seemed a shame when I should have been able to get the info right off the board.
S

RobbyGit, 2025-06-23:
I know , the documents are not very accurate. The pdf and the wiki are out of sync from what i can see. I managed to get it to pulse when it goes past the calibration point.. no thanks to the wiki.

I also notice a few of the serial commands dont behave. I get the feeling they more interested in the cutter and printers than in documenting the mks on its own. I think i need a better closed loop stepper.

lundsimon, 2025-06-24:
Yep, that's where I arrived.