makerbase-mks/MKS-SERVO42C #27: 42C V1.1 on Y Axis Creality 42-40

provalinf, 2022-07-10:
Hi,
I put a Servo42C on an X axis and a 42-34 motor, which works perfectly well.
I put the same module on the Y axis with motor 42-40, but the motor does not move, it just makes a noise at the coils.
When switching to OPEN mode, the motor moves normally with a mA at 1600.

Do you know this problem?
Thanks

mashuptwice, 2023-01-27:
You need to swap the two middle wires of the connector. Use a small screwdriver or a knife to bend back the plastic retainers, carefully pull out the cables and insert them in switched order.

provalinf, 2023-03-19:
Yes, that's what I did, knowing that the same operation worked on the Creality Nema motor of the X axis.  
Don't look for it, the board burned out, I bought a new one with the MKS Nema motor, everything works now.