makerbase-mks/MKS-SERVO42C #29: Firmware V1.2

bodyhey, 2022-08-07:
Dear MKS-Team,
is there already a release date for the SERVO42C firmware version 1.2? There are two issues with v1.1, that should urgently be fixed:
- The problem with the motor to start spinning fast when it is "in a low gear" and some resistance is applied to the shaft.
- The possibility to deactivate the coils via UART: The shaft should be rotating freely, but the encoder values should still be updated.
All the best, 
Martin

ttielemans, 2023-05-18:
I am also looking for the option:
- The possibility to deactivate the coils via UART: The shaft should be rotating freely, but the encoder values should still be updated.
Realization of this option would help realizing my project.

Mpenizzotto, 2025-01-22:
Hi @bodyhey !
I noticed the same problem of motor spinning fast in one direction when some ressistance is apllied to the shaft. I can't find any information of this problem. Could you tell me anything you know about it? Things like when exactly does this happen, or how to avoid it. Thanks a lot!