makerbase-mks/MKS-SERVO42C #30: Motor driver burned

LMDaniel999, 2022-08-15:
Hello,
I have some of these motors in operation on linear axes. Unfortunately, they fail from time to time.

The motor driver burns out and causes a short circuit. Is there any protection for this?

How can I set the maximum current in the "CR_vFOC" mode? The motor can only handle 1A.
I would also like to use this as a torque limiter.

I use an ACC setting of 664. What does this function do exactly? Can this cause the problem?

Changed settings:
Ma --> 400
Protect --> enable
ACC --> 664

Thanks a lot

mozg4D, 2022-12-21:
Same, 4 drivers burned, literally (transistors near 4 output pins on both sides) Ma - 3000, ACC - 1024 vFOC

ThaaSolon, 2023-02-23:
It says in the manual: 

Note: The CR_vFOC and CR_UART mode will automatically adjust the current according to the load.

So I guess you cannot adjust the Ma setting in that mode.