makerbase-mks/MKS-SERVO42C #31: Motor Movement by UART stops on next Command

dcolcott, 2022-08-28:
Hi There,

I've been developing a UART Serial driver on ESP32 / C++ for the Servo42C for a robotic project and found if you send a command to move the motor by a number of steps, this is cancelled by the next UART message regardless of if it is completed or not. 

Example: 
Send a UART command to trigger 30 revolutions (200 pulses per rev x 16 uStepping x 30 = 96000 steps) of the motor at 20 RPM which is expected to take 10 seconds. Once the motor starts moving I want to enter a loop to send UART commands to read the encoder value / received steps to track action and trigger a notification that the motor movement is complete and read to start the next movement.

However, as soon as the SERVO42 receives the proceeding UART command to read the encoder value, the motor stops without completing the full number of requested steps. This makes its it difficult to send a sequence of motor movements one after another. Adding to this is there is a specific UART Motor Stop command suggesting that other commands should not stop the motor.

Can you provide any advice on if this is expected and if so, how to trigger an extended motor movement and monitor or notify its completion via UART?

Thanks.

nikitenkoab, 2022-09-07:
Hello, I have the same problem. dcolcott, do you have any solvign inthis case?
Thanks.

foetusmachine, 2022-10-25:
Same problem.

foetusmachine, 2022-11-12:
Hi everyone here. Not sure if you'll get notified of this message, but I'll leave this comment here anyway.

The new manual that was uploaded a couple of weeks ago, now includes the proper command reference. Under the FD command reference, a new status return has been added. I can confirm this does not work with my V1.1 firmware, but the manual actually refers to V1.1.2.

Hoping this will be solved shortly and MKS will actually release the firmware for us to update our drivers.
[image]

xuchengithub, 2023-02-27:
> Not sure if you'll get notified of this message, but I'll leave this comment here anyway.
i have same issue, but i find use command to move motor use speed control,can get encode position ,so i think we maybe should wirite position controller by our server.

alexisrosa, 2023-09-08:
I just opened an issue before reading this one with the same problem. As soon as I query for the shaft angle, motion is stopped.