makerbase-mks/MKS-SERVO42C #42: Stepper only moves when dirPin is HIGH

ThomasLarang, 2023-02-01:
As the title is described, the stepper only moves when dirPin is set to high, resulting in stepper only being able to move in one direction. What is the cause of this?

imruf84, 2023-02-06:
I have the same issue too.

mashuptwice, 2023-02-07:
How do you set the pin to low? Do you pull it to ground or leave it floating?

imruf84, 2023-02-07:
I'm using this simple code to drive the stepper motor with a simple wiring:
[servo_test.txt](https://github.com/makerbase-mks/MKS-SERVO42C/files/10672286/servo_test.txt)

ThomasLarang, 2023-02-07:
I tried both setting an I/O pin to low, and also tried pull it to the ground. This seems to stop the stepper instead of reversing direction. 

I can change direction by changing dir from CW to CCW in the display menu, but still all movements are stopped when I push dirPin to low.