makerbase-mks/MKS-SERVO42C #53: Motorrunner

MichaelFize, 2023-05-17:
I am currently evaluating the motor (version 1.1.2) for my application.  It runs from 24 V supply with step/dir interface, with step subdivision set to 8.
At the speed of 500 rpm (and probably higher) and under some load, when I stop sending step pulses the motor continues to run for some time. The more time the motor runs, the more the time "overrun". It seems the motor runs at a speed a little slower than it needs, and the position error is accumulated infinitely. However, it doesn't happen at this speed without a load.
I didn't try to play with PID parameters (particularly with Ki), maybe it will help. But with the default parameters it behaves like this, which is unacceptable for my application.

PanosPetrou, 2023-05-23:
I have the same problem too. I can't use the motor reliably because of this

PanosPetrou, 2023-05-26:
If you are using Marlin, check issue #54 it might help.