makerbase-mks/MKS-SERVO42C #54: Motor miscalculating steps causes excess movement.

PanosPetrou, 2023-05-26:
I built a 3D printer recently and I used two MKS SERVO42C motors for the X & Y axes. The printer is using an MKS TinyBee board and running the latest Marlin 2.1.x bugfix. I noticed the motors were miscalculating steps (especially on high speeds) and they were moving more than they should. This was causing all sorts of problems, layer shifts, the X & Y carriages were moving past the usable area and hitting at the end, etc. I suspect it's a firmware issue.

I managed to find a workaround:
On the motor's menu, set the the following settings:
Mode: CR_vFOC
MPlyer: Disable
ACC: Disable

Then go to the motor's menu again and click CAL to calibrate it.

On Marlin's Configuration.h set the following:
#define X_DRIVER_TYPE  DRV8825
#define Y_DRIVER_TYPE  DRV8825
Disable S_CURVE_ACCELERATION (//#define S_CURVE_ACCELERATION) if it's enabled.

On Marlin's Configuration_adv.h set the following:
Disable ADAPTIVE_STEP_SMOOTHING (//#define ADAPTIVE_STEP_SMOOTHING) if it's enabled.

Save changes, compile Marlin and upload it to your printer. The motors should work reliably now.

PS: Makerbase, please fix your firmware, come on, you can do better than that.

PanosPetrou, 2024-04-15:
I had some layer shifts from time to time but now I think I found a workaround once and for all.
1. Go to the motor's menu, restore default settings and click CAL to calibrate it.

2. On Marlin's Configuration.h set the following:
#define X_DRIVER_TYPE  A4988
#define Y_DRIVER_TYPE  A4988
Disable S_CURVE_ACCELERATION (//#define S_CURVE_ACCELERATION) if it's enabled.

3. On Marlin's Configuration_adv.h set the following:
Disable ADAPTIVE_STEP_SMOOTHING (//#define ADAPTIVE_STEP_SMOOTHING) if it's enabled.
Uncomment/set the following:
#define MIN_STEPS_PER_SEGMENT 12
#define MINIMUM_STEPPER_POST_DIR_DELAY 1000
#define MINIMUM_STEPPER_PRE_DIR_DELAY 1000
#define MINIMUM_STEPPER_PULSE 4
#define MAXIMUM_STEPPER_RATE 20000

4. Save changes, compile Marlin and upload it to your printer. The motors should work reliably now.

PS: Makerbase, please fix your firmware, or at least let you customers know that your motor controller can't handle rates over 20KHz well.

closed 2024-04-15 by PanosPetrou

emileballard, 2024-04-16:
as i stated on my post the moment i set ENABLE to HOLD all issues where gone.
i think my problem is because of ESD if i set ENABLE to HIGH and i drag a metal object past the motor i get allot of crakling sounds and sometimes it jumps a huge amount. and offcourse this also happens when i print and dont touch the motor it just jumps to another position. the crackling only happens when i touch the motor with a metal object (allen key)
the moment i set it to HOLD its stable and no more crakling sounds.
for the last 8 months i have had no issues. and my prints look amazing. set to default A4988

PanosPetrou, 2024-04-18:
Good for you. In my case holding the enable pin improved nothing and setting the driver type to A4988 helped slightly but not much. The other settings I mentioned helped a great deal. What you describe as ESD could be an intrerference or grounding issue. The controller of the motor is pretty much exposed which makes it possibly more susceptible to interference.

ste94pz, 2024-05-20:
Hi, do you know the Klipper-equivalent settings? Thank you.

PanosPetrou, 2024-05-21:
> Hi, do you know the Klipper-equivalent settings? Thank you.

I'm not familiar with Klipper unfortunately