makerbase-mks/MKS-SERVO42C #57: Querying for angle while motor moving interrupts motion.

alexisrosa, 2023-09-08:
Hello, 

I am writing a library for a project that moves e few stepper motors with an Arduino using UART serial comm with the MKS-SERVO42C. The motors are connected to the Serial port on the Arduino Nano Every. 

Both motors are connected in a serial network where both motors' rx pins are connected together to the tx pin on the Arduino, and tx pins on the motors are connected together to the rx pin on the Arduino. 

I am able to move, stop, query shaft angle, query encoder values, and query shaft angle error successfully on both motors. 

The issue is that as soon as I try to query for shaft angle while the motor is moving, the motor stops.  I am using the feed for this angle to determine if the motor is still moving before issuing another command. 

Is this normal behavior, or is it something wrong or missing on my end? I have a logic analyzer to monitor all commands received and transmitted. I can see all the commands coming in and going out without problems. 

Thanks in advance.

DanielMarcato, 2024-01-05:
Having the same question but my guess is that this is a feature - not a bug. The controller seems to not be able to handle both in parallel - you can either query the status or have the motor running - not both simultaneously

felixkuebler, 2024-10-05:
I currenty have the same issue. There should be a way to live stream the currenty position. Does anyone have a workaround?

felixkuebler, 2024-10-05:
> Hello,
> 
> I am writing a library for a project that moves e few stepper motors with an Arduino using UART serial comm with the MKS-SERVO42C. The motors are connected to the Serial port 1 on the Arduino Nano Every.
> 
> Both motors are connected in a serial network where both motors' rx pins are connected together to the tx pin on the Arduino, and tx pins on the motors are connected together to the rx pin on the Arduino.
> 
> I am able to move, stop, query shaft angle, query encoder values, and query shaft angle error successfully on both motors.
> 
> The issue is that as soon as I try to query for shaft angle while the motor is moving, the motor stops. I am using the feed for this angle to determine if the motor is still moving before issuing another command.
> 
> Is this normal behavior, or is it something wrong or missing on my end? I have a logic analyzer to monitor all commands received and transmitted. I can see all the commands coming in and going out without problems.
> 
> Thanks in advance.

Did you find a fix?

felixkuebler, 2024-10-05:
#31 is the same issue. I think this one can be closed.

DanielMarcato, 2024-10-05:
Good luck getting anybody to actually take care of this repo 😅