makerbase-mks/MKS-SERVO42C #58: In Serial Uart mode, the command is executed well, but repose is not returned.

homoaroma, 2023-09-18:
For example, if I send the command "e0 f6 10 e6" through serial communication, the motor operates normally.
However, it does not return the “E0 01 E1” response.
I checked the communication line with an oscilloscope, but the board is not sending a response.
Therefore, it is not possible to read the current value of the encoder or the angle of the shaft.
Has anyone had the same experience as me?
I tried changing the board, but the same symptom appears.

alansrobotlab2, 2025-04-21:
fwiw it's quite possible that you're on 1.0 firmware and not 1.1 firmware.  on 1.0 firmware the command bytes are just [0xE0, 0xF6, 0x10]  no checksum byte on 1.0 firmware.

Alexander-Mundy, 2025-10-03:
@homoaroma 
Did you get this figured out?
I just purchased one with V1.1 and the only response I can get after a command is 00.