makerbase-mks/MKS-SERVO42C #62: UART communication protocol does not set 256 microsteps as it only uses a byte

DanielMarcato, 2024-01-18:
The firmware is not capable of setting the microsteppings to 256 via UART even though the display shows 256 as option and this can be chosen via the physical buttons.
Reason is the communication procotol uses a byte for the microstepping and a byte can only have 255 as maximum value.
This is also true for the official tool [MKS SERVO42C CONTROL TOOL](https://github.com/makerbase-mks/MKS-SERVO42C/tree/MKS-SERVO42C-V1.1/MKS%20SERVO42C%20CONTROL%20TOOL) and can be seen in their source code.

kyle-wendling, 2024-02-12:
Ugh, I was considering getting some of these, this type of thing is horrendous, it's such a dumb bug it means that MKS didn't even try their product ONCE before shipping it.

alansrobotlab2, 2025-04-21:
fwiw for anyone wondering, use byte 0x00 to set 256 microsteps.  I saw it in some version of their documentation somewhere.