makerbase-mks/MKS-SERVO42C #64: 'Reverse lookup Error' when doing calibeation

wildblue, 2024-03-02:
What is wrong with my device? I would take a look into the sourcecode (for what GitHub is all about - sourcecodes) myself, but you have first to publish it here.

With the misfittech stepper controller you got this technology for free, please abide by our rules and publish the code of this derived work.

And tell me what's wrong with my device!

Crito02, 2024-03-02:
Just had this issue myself. It should read "hookup" maybe. Try swapping the two middle wires of the lead to the motor. That fixed mine anyway.  Good luck

bart-prog, 2024-08-11:
solved mine 2 thank u for the help

brooin8, 2024-11-25:
This did not help me.  Motor turns both ways during calibration however keeps throwing the error.  Hooked up the uart and did a factory reset and still the problem persists.  I have also verified that it is reading the magnet on startup.  After startup however the readings of the encoder in the uart stay either zero or 220.  Never changing as I move the shaft.  Not sure what is going on as the motor does spin both ways during calibration.  My other 3 work perfect.  Faulty board I assume as I swapped the board with 2 different motors and still the same outcome.  Motors are working motors Mens 17.

kompiz, 2026-05-10:
> This did not help me. Motor turns both ways during calibration however keeps throwing the error. Hooked up the uart and did a factory reset and still the problem persists. I have also verified that [...]

Hi, don't suppose you figured out what was wrong here? Thanks!