makerbase-mks/MKS-SERVO42C #66: Servo42c freezes if EN pin is connected to the main board

Topotopot, 2024-04-24:
hi, I accidentally connected the 6-wire connector backwards. After I discovered this and connected the connector correctly, the controller behaves strangely: if the EN pin is connected to it, the controller freezes immediately after booting (at the ANGLE/ERR/CLK screen). 

If I disconnect the EN pin from the board, the controller works as expected, but I have to set the EN parameters to "hold". 

Here's how the Main board -- Controller were temporarily connected: 
24V -> DIR
GND –> STP 
5V -> EN
EN -> Vkom 
STP -> GND 
DIR -> VCC

I checked the circuit and it looks like the problem may be with the controller's port PB07,  optocouple U8 EL357N or Q1 2N7009. Any ideas how to fix this? Did I burn PB07?

Thank you

PanosPetrou, 2024-05-21:
Something fried, replace these two components if you can.

Topotopot, 2024-05-21:
> Something fried, replace these two components if you can.

Hello, it looks like the problem is in the program, not the hardware.

 I don't recommend buying this controller from Aliexpress because several new controllers bought there are also malfunctioning.

PanosPetrou, 2024-05-21:
Motor hanging at power on, and needing the EN pin disconnected seems like a hardware issue.

Topotopot, 2024-05-21:
> Motor hanging at power on, and needing the EN pin disconnected seems like a hardware issue.

It happens randomly. Sometimes the controller loads and works correctly, but most of the times it freezes in a second  after loading.

PanosPetrou, 2024-05-22:
Then I would check for intermittent connections, short circuits or cold joints on the pcb

Topotopot, 2024-05-22:
> Then I would check for intermittent connections, short circuits or cold joints on the pcb

Looking at reviews on Ali, I am not the one who faced that problem so it seems the problem is with controller itself. I tried it on a CNC and 3d printer, same results. Common drivers work correctly.

PanosPetrou, 2024-06-03:
I don't doubt there is a hardware issue with your motor's board. I'm just saying it either happened because you connected the cable plug wrong, or the board had a problem from the beginning. You said it yourself in your first post: " I accidentally connected the 6-wire connector backwards. After I discovered this and connected the connector correctly, the controller behaves strangely". Personally I have three SERVO42C, all working consistently, they just need a lower frequency signal (at least on my printer), so I set them up as TB6560 in Marlin.